rsmoorthy/ishapi #2: Navigation keys not working

vineshsn, 2018-12-09:
Suppose I disable the mouse pointer in the Add source page of the videos, while the mouse pointer is not pointed to any of the options displayed for selection like <None>, Browse, Add, Remove, OK, Cancel or at the place where I can type a media source..
Then the navigation buttons doesn't work.. I mean it doesn't start navigating from any of the options available for selection...

rsmoorthy, 2019-02-04:
This is solved in 4th Feb image

By doing chmod a+rw /dev/tty0 and uinput

Also, ensure that evdev is added in /etc/modules

rsmoorthy, 2019-02-05:
Not clear or sure

rsmoorthy, 2019-02-07:
Please check if this happens still

vineshsn, 2019-02-08:
It is still happening like that which I can see in the home screen itself.

rsmoorthy, 2019-02-15:
I have tested this explicitly multiple times. Not happening at all.

Based on the other problem (#1 ) -- which also I don't see -- this could be a air mouse specific problem. Please send me that air mouse to see if I can look for work-around. Or suggest how can I test this?

vineshsn, 2019-07-08:
I think this will get resolved with patch 20190709 with keymaping solution fixed for mouse pointer

closed 2019-07-08 by vineshsn