rsmoorthy/ishapi #18: Keymaps folder update didn't work

vineshsn, 2018-12-25:
The 2 .xml files in the program zip file didn't work which worked in the previous version.

https://forum.kodi.tv/showthread.php?tid=203623

Pranam

rsmoorthy, 2019-02-07:
No need for this change. I also checked that if you press that mouse pointer key -- there is a key press coming to Pi when you disable it, but not when enabling it.

vineshsn, 2019-06-12:
I didn't know how to reopen

rsmoorthy, 2019-07-06:
I wish I had paid more attention to this. Actually, I used this keymap exactly and it is working fine. I don't know why it did not work for you in the raspbian version. We simply had to do this only -- perhaps only for Ali TV Remote, which thanks to the new plugin, we are doing it now.

Vinesh, I am terribly sorry for such a long delay in solving this. I should have paid more attention to your solution. If I had, all these time and energy would not have got wasted.

rsmoorthy, 2019-07-06:
Fixing this with the patch on Tuesday

closed 2019-07-18 by vineshsn