rsmoorthy/ishapi #19: Full screen for Music files play

vineshsn, 2018-12-25:
On the selection of a music file let the full screen window open immediately with time line and control buttons for the user to view. Pls ensure the screen saver is enabled after the default 15 seconds which is currently not happening while playing the audio file. Screensaver calling function need not work only for videos...

vineshsn, 2019-02-08:
The screen saver gets called at the beginning and after play end properly sometimes. Not properly sometimes.

vineshsn, 2019-03-19:
It's happening after about 20 to 30 seconds once a music for is played.

vineshsn, 2019-06-11:
This is actually not fixed... The above observation didn't continue to happen. So the solution given by Thomas needs to be updated as patch update to all the Pi's.

ThomasKinder11, 2019-07-18:
One question to this, why should the user see the timeline and menu buttons when audio started?
Full screen I understand but is the file name and menu necessary?

Fullscreen mode can be enable by adding one more xml file to the skin.

Tried this yesterday and it seemed to work for me:

https://forum.kodi.tv/showthread.php?tid=213351

```html
<window type="dialog" id="3001">
    <allowoverlay>no</allowoverlay>
    <onload condition="!Window.IsActive(visualisation)">FullScreen</onload>
    <visible>Player.HasAudio</visible>
    <controls></controls>
</window>
```

vineshsn, 2019-07-18:
Question 1: We don't need to see the controls and timeline unless we press 'OK' button on the remote

vineshsn, 2019-08-02:
In continuation to my comment on 11th June.. I see that Kodi is getting into full-screen mode after 3 mins... I've checked this multiple times... It happens after 3 mins.