rsmoorthy/ishapi #20: Call screensaver (Black) after the video end or video stopped

vineshsn, 2018-12-25:
Call screensaver (Black) function after the video end or video stopped

rsmoorthy, 2019-02-06:
All the code is now written, so that the screensaver is called after the video end or video stopped.
However, this does not always work effectively. After the video is stopped or ended, kodi locks up for few seconds (8 seconds) for some reason and then only all the screen saver plugins are activated. Don't know what it is, will need to check this later.

vineshsn, 2019-02-08:
This is working sometime. Not working many times. But the delay to call the blank screensaver is greatly reduced and it's fine.

rsmoorthy, 2019-02-15:
Related to #30. It was fixed earlier on 07th Feb image, but then closed (revoked) based on #30

closed 2019-02-15 by rsmoorthy

rsmoorthy, 2019-04-03:
This is properly fixed in Pi patch 2019040401 and app version 1.0.10.

Now, we again, call the blank screensaver after play is stopped, play ended and play is paused. There is no delay now.