rsmoorthy/ishapi #22: Linux Pendrive for quick encryption from Laptop or PC

vineshsn, 2018-12-25:
Anna,
We'll need a Linux Pendrive for quick encryption from Laptop or PC unlike using the Raspberry Pi for encryption on the Micro SD card using WinSCP or AndFTP app...
Pranam

vineshsn, 2019-05-19:
Virtualbox ISO image for laptop users and encryption which will give access for local Hard drive of the host machine to create encrypted volumes

rsmoorthy, 2020-03-15:
Completed some time back

closed 2020-03-15 by rsmoorthy

vineshsn, 2020-03-15:
Illa... Adu velai seyyala.... Instead we shifted copying the files using
the terminal kind of app to copy the unencrypted files.

rsmoorthy, 2020-03-15:
If it is not working, please follow up and get it fixed. It is WORKING for me. 
But I realise that you don't want to use this feature any more, after you found another way.