rsmoorthy/ishapi #28: Video controls without black background on the top and bottom

vineshsn, 2019-02-06:
If possible in the final release... While the video is playing when we choose to pause or stop, if the controls can come up without any black background it'll be very good.

ThomasKinder11, 2019-07-17:
I worked on this.  I have simplified the OSD for Video and audio,  removed the animations and unnecessary options in the menu we do not needed,  changed the color of OSD a bit so it's not so visible when it pops up.  I will provide the files here   as soon as find some time.  Then you can test it and see,  and we can adjust if needed.

rsmoorthy, 2019-07-17:
Thanks @ThomasKinder11. Look forward to your files.

ThomasKinder11, 2019-08-12:
https://gitlab.com/thkinder/skin.isha.devel

I have uploaded my skin changes as an new skin to my gitlab. The master branch contains the simplified on screen display for audio and video,  also #34  has been integrated but this is not working properly right now. It sometimes shows two mute symbols on the screen.  When #34 is fixed  I will let you know. 

This is a copy of the Isha Skin with the changes so that we can install it separately for testing.

rsmoorthy, 2019-08-13:
Thank you Thomas for your efforts. I would appreciate if you can raise a PR on github.com, so it is easy to merge.
Also, please  create multiple PRs -- one each for specific fix.

ThomasKinder11, 2019-08-13:
I have never really used git with many people mostly for myself. Just one
question for the pull request, would I need to fork/clone your repo that it
will have the same structure as your repo or is it fine if I upload it as a
separate git repo?

ThomasKinder11, 2019-08-19:
#73 Issued pull request to include development skin into repository for testing where OSD are simplified without animations