rsmoorthy/ishapi #30: Screen saver program is being called during video Play

vineshsn, 2019-02-08:
Once a video starts the screensaver is called within 15 to 20 seconds. i.e., the screen goes black but the audio is alive. By pressing any of the keys on the remote it gets back to the video...

vineshsn, 2019-02-08:
Sometimes after video Play, to revert back to the file list or any operation there is lot of delay from the key press.

vineshsn, 2019-02-08:
Checked with another few files... It's continuing to happen.

vineshsn, 2019-02-08:
I notice it is also being called while navigating through the browsing window of any file section.

vineshsn, 2019-02-13:
Temporarily this feature is disabled for video stopped and video ends functions as this is the cause of the issue at the moment.

rsmoorthy, 2019-02-15:
This feature of calling screensaver after play stop and play ended is being disabled permanently. The main reason is that kodi is not giving control to our plugins, for several seconds after a play stops / ends. We don't know why.

rsmoorthy, 2019-03-21:
We know why kodi is not giving control back to the plugins and this is fixed now. This was due to a "xbmc.sleep(30)" function used within the Resumer plugin. For some reasons, this is not working okay and I changed it to "time.sleep(30)" and Kodi is back again responsive enough.

Kodi is (still) not responsive enough after a) Seek b) Stop c) Pause d) Play ended  -- due to this bug and after this fix, they will be all responsive.

So, we will enable screensaver after play stop and play ended. And play paused too...

rsmoorthy, 2019-04-03:
This is fixed with Pi patch version 2019040401 and App 1.0.10

closed 2019-04-03 by rsmoorthy