rsmoorthy/ishapi #34: Mute indicator disappears

vineshsn, 2019-02-13:
If the indication on mute enabled is retained in the screen it'll be better for people to know that is enabled. Otherwise in the current seeing it's disappearing after a second or two.

rsmoorthy, 2019-02-15:
I have checked in my system, where the mute indicator exists for several hours (not rebooted).  It appears on the right top always and on some screens, at the left bottom also.

I don't understand why is the behavior different to you.

vineshsn, 2019-02-15:
I discovered this while playing the video.. I didn't test during music. Yes
your right about right corner mute icon notification.. But I think that's
while not playing a video or music file...

Pranam

rsmoorthy, 2019-02-15:
I understand your requirement now clearly -- earlier it was not stated correctly.

Before I go and find out how to fix this, can you tell me if this is really needed? I don't want to do anything for meeting "Vinesh's" specific needs, rather it should be for Isha's teacher's genuine needs. These are standard Kodi defaults, where they don't show the mute icon while playing (they fade it out after 5 seconds) - perhaps they might have a reason for it or not, I am not sure.

I want you to understand this one thing clearly -- we have done several customisation now and if we have to upgrade to the next version of Kodi (18), there is lot of work needed, to ensure our customisation is taken forward.

(Though it was difficult and perhaps it is still not fully complete, I understood the need for short screen saver interval and why it is helpful for the teachers. So I did spend time for that).

So, can you tell me if this is really required? If this is indeed required, then I will change it. I have identified the place and the change is simple.

vineshsn, 2019-02-15:
In the previous version it used to be visible on the top right corner all
the time whether we are playing a video or music file. Just because the
mute button is next to the mouse pointer disable button I've received calls
from Teachers in the night saying audio is not working. Thinking and
checking all the reasons it would finally be this mute button enabled.
Sometimes they don't recognise this also because the screen resolution on
the projector screen. So kindly fix this Anna...

Pranam

rsmoorthy, 2019-03-21:
Though I said, this is easy to fix -- I have not been able to fix this. Will keep this open, until I am able to fix this.

ThomasKinder11, 2019-07-18:
In my version the mute symbol is shown in the main menu a the time but not in the screensaver,  or full screen Video and Music windows. 

With a short test I was able to display it when screensaver is on,  or when Music Fullscreen is enabled.  

The change is a few lines of code in some Xml files.  I need to identify all the xml that need to be changed but should not be to complicated.  Will show what needs to be changed when I implemented and tested it properly.