rsmoorthy/ishapi #36: Music folder in Rootfs to be given permission

vineshsn, 2019-02-14:
The Music folder in Rootfs to be given permission to add files and play music files as the volunteers would not be able to play it until the teacher brings their mobile to authenticate and Mount the volume.

rsmoorthy, 2019-02-15:
Can you create one or more directories under /mnt/Ishamedia and place unencrypted files directly under that? This way, users can just go to "Ishamedia" folder and see all their files there. Unencrypted files will always be visible, whether teacher connects to it or not.

If the above solution is okay, please close the ticket yourself.

vineshsn, 2019-02-15:
This is also fine Anna 🙏
Pranam

closed 2019-02-15 by rsmoorthy

vineshsn, 2019-02-28:
Hey,

I just signed the petition ". @narendramodi @PMOIndia @imrankhanPTI @United
Nations: Release of Indian Wing Commander Abhinandan with honour from
Pakistan" and wanted to see if you could help by adding your name.

Our goal is to reach 25,000 signatures and we need more support. You can
read more and sign the petition here:

http://chng.it/qRsHChSXSd

Thanks!
Vinesh