rsmoorthy/ishapi #38: App corrections

vineshsn, 2019-03-19:
1. Google sign-up to be enabled
2. under sign-up be to remove the 'optional' for mobile number and email ID fields
3. In the home page on clicking the connect it would be good to display the encrypted volume getting mounted one after another for the user to know.
4. OTP via SMS to be enabled
5. Remote control to be fixed (as informed by Moorthy Anna). Thus is just for reminder and everyone's reference.
6. Look and feel of the app can be made better

rsmoorthy, 2019-03-21:
Creating like this won't help close this issue for long time. Ideally better to create unrelated issues as separate issues.

rsmoorthy, 2019-05-11:
This is being fixed in 1.0.14.

closed 2019-05-11 by rsmoorthy