rsmoorthy/ishapi #47: Filter, Search and Region information under users section for Admin

vineshsn, 2019-04-05:
If Search option is made available under users section it'll be helpful to check out update any users credentials. Filter could be used to see users of specific roles or group or region.

The 'Region' management of users will be given under admin action or admin tasks section where if a new region needs to be created it'll be given here (For Ex. AP&T, Western, Eastern, North, South or TN, etc.,). But assigning a pre-created region to every user will be done under users section.

rsmoorthy, 2019-05-11:
Please provide me the list of Regions

vineshsn, 2019-05-12:
North
South
East
West
AP & Telangana
TN-North
TN-South
TN-East
TN-West
Chennai
APAC
UK
US
Middle East
China
Europe

Pranam

rsmoorthy, 2019-07-07:
This is fixed in 1.0.15

rsmoorthy, 2019-07-07:
You can filter in one of the following ways:
a) Simple Filter: Just type few chars of either name / mobile / email. Like just typing 'moo' will list all names or emails or mobile having those characters.

b) Advanced Filter: See examples for illustration
"disabled:Yes"    for all disabled users
"group:group1" for a list of all users in "Group1"
"region:US" for a list of all users in US region
"role:admin" for a list of admin roles (this will return "UserAdmin" roles also)

"group:group2 disabled:Yes"   for all disabled users in Group2
"region:chennai group:group1 mobile:98"  for all users in  region chennai, who are in group1 with mobile number  having 98 in them.

As of now, no negative searches or complex combination of "and" / "or" is possible.

The fields available are: name,mobile,email,region,group,disabled,role

vineshsn, 2019-07-18:
Could you add Ashram as one more region? Is it possible for me to create region in the current version?

closed 2020-03-15 by rsmoorthy