rsmoorthy/ishapi #63: Obtaining IP address error

vineshsn, 2019-06-11:
For a few Android phones and iPhones while connecting to 'ishapi' WiFi network (after giving b the correct password) the phones show 'Obtaining IP address...' and after some time or after making a few attempts it says 'IP Configuration error'.

Similar thing was observed on an iPhone where it showed 'No Internet. Connecting...' and doesn't get connected.

vineshsn, 2019-06-11:
Solution suggested is to try a standard hotspot SSID and password on the Pi which is able to get connected to the Pi and update the Pi... Then try connecting the faulty phone by renaming the hotspot SSID and password of the phone and check if it's working. If it works this can be standardized way of connecting the Pi.

vineshsn, 2019-06-11:
Also check the settings of the phone if it's unable to connect because there is no internet connectivity on the Pi.

vineshsn, 2019-06-11:
Solution suggested by Moorthy Anna for long term, mass demonstration and training is to make the SSID of every Pi unique like 'ishapixxxxxx' were xxxxxx is the last 3 numbers of Mac IP address.

rsmoorthy, 2019-07-07:
Two different problems, combined here.
a) Obtaining IP address error: This is for now, assumed to be, because of the mobile doing some internet check and if the "Internet" is not available, not continuing with the connection. This is too bad, if this is so. We are now behaving as if we are "public wifi" to fix this.

b) Differentiate between two IshaPi: This is now done with unique SSID for each Pi

closed 2019-07-18 by vineshsn