rsmoorthy/ishapi #68: Error message from video resume after seeking video

ThomasKinder11, 2019-07-31:
I get the following error message after I was seeking a video, and the video ended. Seeking was done by entering a number greater then the video time while video was playing. 

@rsmoorthy  I was wondering if we actually need this plugin. My test have shown that kodi itslef also saves the last view point in the kodi database for a video isn't it`? I did a few test, playing a video, checking the playing time and then just unplugged the power to the pi. Then when I rebooted and selected the same video again it asked me if I want to restart from the last point or from beginning. The timings from the last point where maximum 20 seconds before I unplugged the power.  In the past when the PI was restarted the video resumer plugin was automatically starting the last video, but this does not work anymore as we need to login with py app first before video are available. With Kodi internal implementation the user would manually need to select the last file but the restart point should be set properly, I have tested this 5 times and the restart point of kodi was always correct. 

See also issue for advancedsettings.xml file regardind restart points #35 

**Error message:**

`````bash
callback/script returned the following error<--- NOTE: IGNORING THIS CAN LEAD TO MEMORY LEAKS!

Error Type: <type 'exceptions.RuntimeError'>
Error Contents: XBMC is not playing any file
Traceback (most recent call last):
File "/usr/share/kodi/addons/plugin.video.resumer/default.py", line 207, in onPlayBackSeek self._do_update()
File "/usr/share/kodi/addons/plugin.video.resumer/default.py", line 179, in _do_update
filename = xbmc.Player().getPlayingFile()
RuntimeError: XBMC is not playing any file
-->End of Python script error report<--
```

rsmoorthy, 2019-08-09:
@ThomasKinder11 Did you do the above test with this plugin disabled or enabled? By default, the Kodi  does not save the checkpoints so frequently (every 30s). If I am correct, your experiments have worked because of this plugin. Correct me if I am wrong and your experiments have been done with the plugin disabled -- in which case, we can remove this plugin.

rsmoorthy, 2019-08-09:
I will take a look at this error message  and also #35. I think this #35 was there because of some other issue in video resumer, which was long back fixed. In any case, let me take a fresh take look at it.

ThomasKinder11, 2019-08-09:
I have to check if I tested it with the video resumer enable or not. 
I will double check maybe it is just my misunderstanding of how things work. But the error message should be remove if possible.