rsmoorthy/ishapi #69: Audio and Video out of sync

ThomasKinder11, 2019-07-31:
Myself and I know of at least two more Ishangas who have the same problem. Sometimes the video and audio are not in sync. Audio comes 250ms-500ms faster then video. So far I was not able to find out what the reason for this is. It does not happen anytime, but in last class almost every video I faced this problem. But sometimes after restart the problem is gone. 

What I have tried now is to experiment with the video buffer settings. 

````xml
<advancedsettings>
    <buffermode>1</buffermode>
</advancedsettings>
````

buffermode = 1 : will enable video buffering for all video files, regardless of weather they are stored on local disk or coming via streaming sources. I will test in the next class if this problem still accours. The few test I have done did not show the audio delay. 

I am in London center  and had big problems with this in the last class. Almost every video had this issue. As I am using the same setup I hope to be able to identify the issue.

ThomasKinder11, 2019-08-03:
This seeting change did not solve the problem still getting video and audio out of sync for many videos .  Now I will try to add <readfactor>1000</readfactor> to the advanced setting  as well and see if this makes any difference

rsmoorthy, 2019-08-09:
I think it will be more helpful, if we can simulate this or identify if it happens.

ThomasKinder11, 2019-08-09:
I have not found a way to replicate this consistently. When I play the files after class the problem most often is not there, even when I restart the PI its not there, its very random so not able to trigger this problem consistently to analyze the issue.