rsmoorthy/ishapi #70: CEC Adapter error in Kodi log file

ThomasKinder11, 2019-07-31:
I see the follwing error message in Kodi log. When I disable the CEC adapter under System - Settings - Input - Peripherals - CEC adapter messages dissaper. I have no monitor connected to PI and use VLC. Is that the problem that no HDMI connection is available?

Do we need to fix this issue or can we disable CEC?

ERROR: CecLogMessage - could not open a connection (try  5)                                                                                   09:50:36.887 T:1739584256   ERROR: CecLogMessage - RegisterLogicalAddress - CEC is b eing used by another application. Run "tvservice --off" and try again.               09:50:36.887 T:1739584256   ERROR: CecLogMessage - Open - vc_cec could not be initia lised                                                                                09:50:36.887 T:1739584256   ERROR: CecLogMessage - could not open a connection (try  6)                                                                                   09:50:37.888 T:1739584256   ERROR: CecLogMessage - RegisterLogicalAddress - CEC is b eing used by another application. Run "tvservice --off" and try again.               09:50:37.888 T:1739584256   ERROR: CecLogMessage - Open - vc_cec could not be initia lised

rsmoorthy, 2019-08-09:
@ThomasKinder11 - Yes Thomas, I too see these messages, when not connected to my TV. It is annoying, while developing and so I go and manually disable CEC.

I don't want to disable CEC, if it can be effectively used by folks - who like to use that.

This is not really an issue. If CEC is not there, there is extra log messages (they can send it less frequently) -- indicating that it is not able to find a CEC device.

I will close this, if you agree with me.

ThomasKinder11, 2019-08-09:
Yes please close, its only happening with  vnc if HDMI is connected nor problem so its fine.

closed 2019-08-09 by ThomasKinder11