rsmoorthy/ishapi #72: USB memory sticks with exfat not mounted

ThomasKinder11, 2019-08-08:
I recently bought a new memory stick which is exfat formatted. File system support for exfat is not installed. 

I  installed two packets
Exfat-fuse._1.2.5-2
Exfat-utils_1.2.5.2

Now I can manually mount the USB memory stick with SSH but if usbmount tries to mount it I get the following error:

"Transport Endpoint is not connected"

I edited usbmount.conf to add exfat to the supported files systems. 

I know of another teacher who had to use a laptop because USB drive was also not mounted, but I do not know if it is the same issue with file system or something else.

rsmoorthy, 2019-08-09:
Ok thanks. I will add exfat support in the 2.0 version.

ThomasKinder11, 2019-08-09:
I had some problems with usbmount. 
here is one thing that I found because the people say usbmount is not compatible with exfat
https://serverfault.com/questions/766506/automount-usb-drives-with-systemd/767079#767079

Because I am not so good in bash scripting I have tried this approach with python. I have uploaded everything in my goodle drive. Maybe @rsmoorthy  you have a better way of implementing this. 

Basically it uses systemd and udev rule for mounting the memory devices. 

https://drive.google.com/open?id=1FXNkgIF07Jrj9ub7NRzvbG_P5-3LH_LG

(I copied the Zip file to FAT filesystem so owner and access of the files would need to be changed under Linux....)

rsmoorthy, 2020-03-15:
Fixed in some 2.0.x version. Yes, I had to fix some code of usbmount (using bash scripts)

closed 2020-03-15 by rsmoorthy