rsmoorthy/ishapi #75: PI Wifi Patching Process

ThomasKinder11, 2019-08-29:
One of the Ishangas recently did an update on the Pi. After rebooting a few things in the Kodi Gui did not work anymore. I was not able to check the SD card myself and do not know exactly what went wrong but I had a look into the piserver and the app source code to check weather we are using some kind of checking/restore mechanism if something went wrong. 

@rsmoorthy please correct me if its wrong what I am writing.

In a broad view the update process works like this right now:
1. When an update is available the PI will download the patch file from the server
2. Then it will take one file after another (function sendPiPatch() in UserHome.js: line 171) and sends the content of the file via post request to the piserver
3. The pi server (web.py line 250) receives the packet and saves the content of the packet into the specified file. If the file already exist, old file content is overridden. This basically replaces already existing files or generates new files if they did not exist before. 

**Point 1:**
What I cannot see is if we use any kind of checking if the file has been written correctly. For example by generating and comparing a md5 hash value. This means if something went wrong in writing the file, or processing of the data we will not be able to detect this and the system might not work properly after update. 
Would it be possible and needed to integrate checksum test to check the files?

**Point 2:**
Files are transferred one by one over the wifi connection and updated in the order they arrive.
This means that if the connection is interrupted in between the data transmission we only partially update the system. If this happens, how does the system detect this and retries the update procedure? Or is there a way to manually force reinstalling of an update in case something went wrong?

rsmoorthy, 2019-08-31:
All excellent points Thomas. Thanks. I appreciate your inputs.
Coincidentally, I was told that 2 people had problems. I spoke to one person on the phone, and found out that the Pi was not sending out IP address. It was working last month. I don't really know what happened, but patching process as the culprit is a possible hypothesis. Though to be verified.

I thought about both your points. Verification by md5 is great, will do it, but won't be practically helpful (solve our real problem). The reason: file changes are still in the RAM cache (OS level) and wont really verify if it has been written to SD card. (Long story, short: This is something I have spent time on linux servers, even in AWS env, we simply have to trust the OS and the hardware).

Files are indeed being sent one by one (for a specific reason - Patching is not just copying files, but also issuing commands, in some specific order). In theory, what you specified is possible. But in practise, this is not an issue. I have been conscious enough, to use only very small files (to be patched) and the whole transaction takes less than a sec. So, yes, in theory, we could have an unstable system - because of some unknown issues in the patching process failure. But practically I don't expect them to happen (due to network, due to SD card issues) -- except for software induced scenarios.

I have thinking some more and then realised what could be the problem (still not 100% sure). We are doing the patch and immediately doing a reboot. What I have missed: running the "sync" command to flush out the OS RAM cache to the disk. This is something that I will do immediately (in the next 1 or 2 days).

On the larger point, I would like to address this problem in 2.0, where if something went wrong - fixing the same. Unlike a database "transaction", we cannot have a rollback on failure at the system level. This is an area, I will fix this in a much more robust way. (Right now, we can "brick" the Pi and have to rewrite the whole SD card).

Also, I have asked the Bangalore center to send me the not-working Pi -- if it happens, I can confirm what really happened.

rsmoorthy, 2019-08-31:
After I read what I wrote above, I realised that this point has not been expressed clearly.

Rephrasing, there is no simple way to really ensure that any failures in the patching process are captured and rolled back effectively. Also, it consumes lot of effort towards this and we get very little gain. Instead, if a failure has happened (either detected by system or not), we will focus on how to fix them (either automatically or manually). This is exactly what you also asked ("If this happens, how does the system detect this and retries the update procedure? Or is there a way to manually force reinstalling of an update in case something went wrong?")

The solution will be more in the lines of an embedded device (like switches, routers), which has a mechanism to do "hard reset" or "factory default". Currently also, people can manually do the "hard reset", by reinstalling the SD card contents. But that is too tedious. We will be having a simple solution to "hard reset" -- without having to overwrite the SD card contents. This will be part of 2.0

rsmoorthy, 2019-10-21:
It looks like the problem of why some people were not able to connect to, turned out to be a completely different problem. Thomas mentioned that "Kodi GUI itself did not come up", but for others, the problem was "IP Configuration Error". Here are the details of that problem:

1. The DHCP server from Pi issues a IP address to a mobile. This info is stored in Pi's local leases file. The lease is given for 24 hours.
2. The Pi does not have a local clock, so the time in the Pi does not move with the "real time". (If you switch off the Pi today and turn on the Pi after 1 month, the time will still be the old time that was 1 month before).
3. So the DHCP lease does not expire so soon as we think.
4. I had also configured only 10 IPs for DHCP. So once all the 10 IPs are issued (even over a period of several days), Pi still thinks it has exhausted the quota and cannot give any more IPs.

For fixing this, I am going to do the following:
1. Set date and time on the Pi every time the mobile connects to the Pi
2. Patch dnsmasq configuration so that it issues 30 IPs instead of just 10
3. Change dnsmasq configuration so that the lease time is reduced to 1h
4. On every reboot of pi, delete the dnsmasq leases

Reg Thomas's problem -- that he observed -- that the Kodi UI itself was not coming up, I really don't know, but I am going to slightly improve the patching process by calling "sync" before calling reboot.

closed 2020-03-15 by rsmoorthy